raeedcho/temper #1: Mill-Max pins part number?

Ramblurr, 2023-09-18:
Hi there! Building a keyboard for the first time, do you have a part/model number for the Mill-Max pins?

calerouxz, 2023-09-18:
I believe the sockets are Mill-Max 315-43-164-41-001000 which you can find on [digikey](https://www.digikey.com/en/products/detail/mill-max-manufacturing-corp/315-43-164-41-001000/1212142).
And you could use pins 3320-0-00-15-00-00-03-0 which you can also find on [digikey](https://www.digikey.com/en/products/detail/mill-max-manufacturing-corp/3320-0-00-15-00-00-03-0/4147392).

I would also like to know about the switches though.

Ramblurr, 2023-09-18:
> I believe the sockets are Mill-Max 315-43-164-41-001000 which you can find on [digikey](https://www.digikey.com/en/products/detail/mill-max-manufacturing-corp/315-43-164-41-001000/1212142).

@raeedcho mentions Mill-Max 310 sockets.. which I think are part number 310-93-132-41-001000 (don't seem to be available on digikey, but are available at mouser)

Ramblurr, 2023-09-18:
Good to know! Since @raeedcho mentioned 510 by name, it's probably the one to get (with 64 positions).

I researched some other keyboard builds and guess that the Alps SPDT switch might be the [SSSS811101](https://tech.alpsalpine.com/e/products/detail/SSSS811101/) , it seems to be a match to the photo.

Ramblurr, 2023-09-18:
Also the Panasonic momentary switch looks like the [EVQ-PUC02K](https://octopart.com/evq-puc02k-panasonic-41161443?r=sp)

calerouxz, 2023-09-18:
Great to know! I'm pretty much gearing up to build one of these in the near future.

Ramblurr, 2023-09-18:
Ok

> The 310 also seems to have a different contact finish, as it's described as 'tin-lead'

I found another 310 that has 64 positions and no lead: 310-43-164-41-001000

calerouxz, 2023-09-18:
Seems correct. The 310 seems to have longer tails, and is probably considered the 'standard' tail socket according to mill-max's website. 
The 315 would probably be a good option if paired with pin headers. This is probably the pairing Typeractive seems to sell as 'ex-solder' sockets and headers.

raeedcho, 2023-09-18:
I sourced many of my parts from [Typeractive](https://typeractive.xyz/collections/parts). There should be information and data sheets there for most of the things.

Specifically:
- Alps SPDT switches part number [SSSS811101](https://cdn.shopify.com/s/files/1/0618/5674/3655/files/ALPS-SSSS811101.pdf?v=1670451309)
- Panasonic momentary switches part number [EVQPUL](https://cdn.shopify.com/s/files/1/0618/5674/3655/files/PANASONIC-EVQPUC02K.pdf?v=1670451309)
- Mill Max pins/sockets I'm not 100% sure of the part numbers, but I got them from [here](https://typeractive.xyz/products/machine-sockets-and-pins)

Hopefully that clears up the confusion, but let me know if it doesn't.

Ramblurr, 2023-09-18:
thanks @raeedcho that clears up a lot! 

I've never worked with kicad project files before when printing PCBs (you've got a real nub on your hands sorry). Is there anything else necessary besides opening the project and doing File > Fabrication Outputs > Gerber to produce an output I can zip up and send to PCBWay?

raeedcho, 2023-09-18:
If you just want the gerbers, I've uploaded them to the releases of this repository [here](https://github.com/raeedcho/temper/releases/tag/pcb-v1.0.0)

Come to think of it, I should probably put that in the documentation for others who might have missed that.

closed 2023-09-18 by raeedcho

triple1c86, 2023-12-12:
Hi, does the keyboard work without the reset button? I have soldered everything including the battery except reset button and nano view. I have also flashed your temper firmware on left split keyboard. I have seen the usb device detected on my arch linux computer as OpenMoko. Inc. Temper, however none of the keys work..

I have made a wired corne keyboard and flashed with qmk. However, I'm quite new with zmk firmware, so I wonder if there's anything that I can troubleshoot this with.. Sorry if this is not the right place to ask about this, however I'll appreciate it if someone can point me the right directions to this.. Tyvm.

raeedcho, 2023-12-12:
The keyboard should work without the reset button (assuming you short the reset pins to flash the boards). I'm not sure what the OpenMoko. Inc. Temper is about--this may be soldering issue or a ZMK firmware issue, I think. Perhaps double check the ZMK firmware settings and look for support in the ZMK GitHub repo or Discord server.

By the way, this thread you've commented in is an unrelated closed issue--it's not a problem, but if you have more questions on this for me, I would appreciate it if you opened a new GitHub issue specifically tailored to this.

Good luck!