raeedcho/temper #2: About securing the bottom plate

purefall0624, 2023-09-18:
Hello! I really loves the minimalist design of your keyboard. But I'm wondering how to secure the pcb with the bottom plate, you know, about the models of the screws and maybe existing studs. Love from China!

raeedcho, 2023-09-18:
Thanks! As I've built it in the pictures, I just used M2 screws and spacers. The screws are 3mm M2 screws, and they're attached to 5mm brass M2 spacers (also known as standoffs sometimes I think) through the PCB. The bottom plate slides over the spacers on the bottom of the PCB and secures to the spacers with the same 3mm M2 screws.

I'd love to see the result if you end up building one!

raeedcho, 2023-09-18:
I realized that I didn't put this information in the README before--it's there now. Feel free to reopen the issue if you have more questions!