raeedcho/temper #3: JST Battery Connector?

calerouxz, 2023-10-03:
Is it possible to use a JST battery connectors like these [typeractive JST connector](https://typeractive.xyz/products/battery-jack) or [the one included with the batteries with connectors](https://typeractive.xyz/products/lithium-battery-110mah?variant=43294731665639) with the current design? Or maybe the pads for the connector could be added in the future?

raeedcho, 2023-10-03:
You can definitely use the pigtail connector with the current pad design--you just have to solder the wires into the right spots.

The current design doesn't support the board-mounted JST connector, and I'm not currently planning on adding those pads, though that could change. I had initially wanted to use the board-mounted connector you linked, but found it slightly difficult to make space on the board for it alongside the power and reset switches. I ended up deciding that it would be easy enough to clip or desolder the battery if I ever needed to replace it, but I definitely see the advantages of having the connector.

calerouxz, 2023-10-03:
I was thinking the pigtail connector could work, I'm a bit worried that it could be too long, making packaging the battery difficult on the board, might be something worth trying regardless.

closed 2023-10-03 by raeedcho