raeedcho/temper #5: Bottom case dimensions

tmcdonell, 2023-11-06:
Hi! Thanks for creating and releasing this board!

I need to order the bottom plate, but when I upload the design to my local cutting service (I tried two) they estimate that the final board will be about 83mm x 67 mm, which is quite a bit smaller than the PCB so doesn't seem right...

I had to convert the .svg to either .pdf or .dxf for these services and used inkscape to do that, which is perhaps where it is going wrong. Do you have any suggestions, or have the file in another format already that I can try? (.ai is also accepted)

raeedcho, 2023-11-08:
Hmm, I'm not sure why Inkscape would change the scaling, but I guess I've also never used Inkscape, so I can't really comment. I tried exporting a DXF from Illustrator, but it seemed to be blank. I can upload a PDF version here though:

[temper-bottomplate.pdf](https://github.com/raeedcho/temper/files/13298602/temper-bottomplate.pdf)

Let me know if this works for you!

tmcdonell, 2023-11-09:
Thanks, that seems to have worked! :D Size of the design is now reported as 111 x 90mm. Ordered and should be here in a week or so.

closed 2023-11-09 by tmcdonell