raeedcho/temper #6: Question about: Solder the jumpers underneath the microcontrollers

triple1c86, 2023-12-26:
Hi, just a question about Solder the jumpers underneath the microcontrollers part:

1. do you mean have to solder some tin on the red circle spot shown in the picture?

then another question is if i want to rule out battery issue, can i just solder a wire on the green circle spot?

I be kindly appreciated if you can help a noob like myself out. Thank you. 
 
[image]

raeedcho, 2023-12-26:
Yes, soldering that jumper you circled basically means that you need to electrically connect the two pads with solder.

As for the battery pads, definitely don't connect those--it could result in a short in the microcontroller.

I'm not sure what you mean by ruling out a battery issue--what issue are you having? You can power the microcontroller without the battery connected by plugging it in to usb, if you want to test your firmware before hooking up the battery.

triple1c86, 2023-12-26:
omg.. wondered what went wrong when none of the buttons work after hours and hours of resoldering this and that, thinking it was my poor soldering skills.. So I revisited your instructions again, only to realise that I read this before and thought its something not important and skipped it.. silly me. thanks for clarifying that.

the battery part was my desparate attempt to fix the none of buttons work issue that I thought its related to..

Most likely thats the solution for me. Thank you very much mate. Ya the best! @raeedcho and thanks for such a wonderful wireless keyboard design!

closed 2023-12-26 by triple1c86