raeedcho/temper #8: STEP file for case design

calerouxz, 2024-01-14:
Hi, 

Would it be possible to get STEP files to be able to design a 3D-printable case specifically for the Temper? I'd like to include something to better handle the on/off switch among other things.

raeedcho, 2024-01-14:
I'm not sure how to generate STEP files for this, to be honest--if you know how, I'd be happy to add them to the repository.

For the most part, the case files for the original chocofi will basically work for the temper as well, though--that might be a good starting place, if you're willing to give it a go.

calerouxz, 2024-01-14:
I'm currently using a Chocofi case on my temper, but the case design in general could use a bit of refinement, I would also like to try my hand at making an easier rocker for the on/off switch, some tenting options, as well as some nice covers for the nice!nano. Unfortunately I couldn't find a .step file in the Chocofi repo either, but I'd still prefer to start from the temper's, as that would give me an accurate placement for the components.

[Here](https://www.pcbway.com/blog/PCB_Design_Tutorial/How_to_export_a_PCB__step__from_KICAD_to_Fusion_360.html) is a tutorial from PCBWay on how to export to a .step file.

raeedcho, 2024-01-15:
I gave it a shot through KiCAD, and I'm putting the file at this link for now--is this what you're after? https://autode.sk/4aUPt1M

As far as I can tell, this is just a STEP file for the PCB by itself, but doesn't have any of the components on it (since it's a reversible board, I'm not sure how I would put the components on it correctly anyway).

calerouxz, 2024-01-15:
This is exactly what I'm after! The components are not particularly important for what I need them for. This just allows me to have an accurate representation of the board while creating the case.

Thank you very much!

raeedcho, 2024-01-15:
Great! Good luck with the case, I'd love to see it when it's done 😁