raeedcho/temper #10: Troubleshooting battery and PCB

keithadams, 2024-03-11:
I'm at the final stage of building a temper and I can't seem to get the battery working on the PCB. I soldered the battery to the PCB and the nice!nano v2 doesn't power on. The nice!nano v2 works when I connect the battery directly to it and the keyboard works, but obviously the power switch can't be used. Any ideas on how I can troubleshoot and isolate the problem? Thanks!

calerouxz, 2024-03-11:
Is it possible to see how you soldered the nicd nano and the battery?

keithadams, 2024-03-12:
I ended up desoldering the battery to test it directly connected to the nice!nano. I had the wires connected like this and I was careful not to accidentally bridge the two terminals.

[image]

When I connected it directly, the battery worked immediately. Does that rule out the battery as a problem? I charged the battery to 100% and tried again with the PCB, but only using heat resistant tape. I was holding off on soldering again until I narrowed down the issue but I'm willing to re-solder. Just trying to figure out the best way to proceed.

raeedcho, 2024-03-13:
Do you have a picture of your actual board? It's possible there's something else wrong that would be easier to see in a picture. For example, have you soldered the jumpers on the correct side of the PCB?

keithadams, 2024-03-13:
Yes, here's the board as it is now. Also, I have a multimeter available to test. 

[image]

raeedcho, 2024-03-14:
Thanks for the picture. Have you tested with the multimeter to see whether the battery terminal pads are connected to the correct socket pins? Check the [nice!nano pinout](https://nicekeyboards.com/docs/nice-nano/pinout-schematic/ for reference, and make sure the power switch is in the "on" state when you check, since the power switch disconnects the battery from the microcontroller.

If it's not connected, check the connection through the power switch pads. The positive pad should be routed to one of the pads on the power switch, and the socket pin corresponding to battery power should be routed to a different power switch pad.

keithadams, 2024-03-25:
Thanks for the replies! I ordered a new battery just to be safe and soldered it to the PCB. Next, I tried to do a continuity test with a multimeter. Please let me know if I did it incorrectly.

[image]

When I connect 1 and 2, there is no beep. The power switch is set to "on".

When I connect 1 and 3, there is a beep, regardless of when the switch is "on" or "off".

raeedcho, 2024-03-25:
Have you tried testing the connection between the pin or pad labeled 2 to the top of the three pads to the left of the power switch? I believe these two should be connected regardless of switch state.

If that connection is not good, then it may be the case that the jumper you soldered on the top is not fully connected (it kind of looks like it from the picture). Test whether the pin you indicated as 2 is connected to the blob of solder labeled "RAW" right next to it.

If the connection from the RAW pin to the power switch is fine, then maybe test whether the top two pads on the left of the power switch are connected when the switch is on and not connected when it's off.

keithadams, 2024-04-02:
The toggle switch was the problem and I was able to fix the issue.

First, I used an extra PCB to do a continuity test with an unsoldered toggle switch and found the switch is a little finicky. It works best when pressure is applied towards the PCB. I decided to remove the switch on my board but unfortunately a pad came off in the process. I started fresh with a new board, but this time I started with the power switch. Here's the process that helped me:

1. Place the toggle switch in position and apply pressure towards the board with tweezers.
2. Solder a **single** pin first, and then test for continuity. This means connecting to the positive battery terminal and the RAW pad closer to the battery terminal pads. 
3. If there is no continuity, then try again by heating up the solder joint and re-seating the toggle switch.
4. Once it works, solder the rest of the pins on the toggle switch.

Thanks again for the guidance!

closed 2024-04-02 by keithadams