raeedcho/temper #11: Case designed and available

calerouxz, 2024-04-08:
@raeedcho , I know it's been over 2 months since I asked for the PCB model, but I've finally been able to create the case. I'm not done with the documentation just yet and it's missing the real life images, but here's the github repo I've set up for it in case you'd like to link it: [Temper gasket case](https://github.com/calerouxz/TemperKB-case)

I made a post on reddit with an image that you can view: [reddit post](https://www.reddit.com/r/ErgoMechKeyboards/comments/1byyiem/i_was_finally_able_to_finish_this_case_for_my/)

raeedcho, 2024-04-10:
Wonderful! I'm out of town at the moment, but I'll add links in the readme when I get a chance. Good work!

raeedcho, 2024-04-13:
Added some links to the readme. Thanks!

closed 2024-04-13 by raeedcho