raeedcho/temper #12: Diodeless?

ThomasAlban, 2024-04-10:
Is there any way this could be modified to be diodeless? I'm definitely not an expert, but it seems like there would be few enough keys on each side for there to be 1 GPIO per key (correct me if I'm wrong!)

calerouxz, 2024-04-12:
I took a look at the schematic. I'm pretty sure the nice!view is taking some of the pin on the nice!nano, so there's no longer enough available pins to run all 18 switches per side without diodes.

raeedcho, 2024-04-13:
Ah, yes, I couldn't remember before, but I think that's the reason I didn't make it diode-less. It could be possible to make another version of the temper without the diodes or the nice!view (or the sockets, for that matter) for a very minimal build, but it's not on my radar at the moment.

closed 2024-04-13 by raeedcho

ThomasAlban, 2024-04-13:
Ah ok makes sense. I might have a go at making this at some point, as I've been wanting to learn more about this anyway. Thanks for your help!