raeedcho/temper #13: Soldering the jumpers - front or back

paradoxicator, 2024-04-14:
Hello - thank you for your excellent work!  I just wanted to clarify the jumpers - I wasn't understanding if the jumpers needed to be soldered on the front or back of the PCB..

I have completed the whole build, but I soldered it on the back (instead of the battery/front side).  After looking at some pictures of prior questions/issue posts, I noticed that there was a picture of the jumpers soldered on the front side.

Please forgive this newbie if this is a stupid question.  I have never built one with jumpers like this before.

raeedcho, 2024-04-16:
Hi! The jumpers need to be soldered on the front side, under the battery. In hindsight, I probably should have designed it the other way, but that's how it is.

If you've already soldered the jumpers but haven't done the rest, there are a few easy fixes that I think might work:

- First, if nothing else is done, then just flip the PCBs around--they're mirror images, so there's no problem
- If you haven't socketed the microcontroller yet, I believe you can socket that "upside-down", if you're okay with that. The downside is that if you then move to another keyboard and use the same microcontroller, you might have to be aware of how to connect things.
- If things are all soldered, you might have no choice but to desolder some things. If you've never desoldered anything before, I suggest getting some braided solder wick and watching some YouTube videos on desoldering.

Good luck!

KastenTaco, 2024-04-16:
Please make adjust the build instructions accordingly.

paradoxicator, 2024-04-16:
I actually completed everything before I realized this.. I soldered it on the correct side and desoldered the back with a solder wick.  Thank you for the clarification!  It's a great board!

AngryTacoEater, 2024-04-16:
It's great indeed! Didn't mean to sound snarky. Just frustrating when this happens. Would be a shame if others have this happen. The Nice!Nano is pricy to burn up.

raeedcho, 2024-04-17:
The instructions are in the readme already--the first instruction actually:

> - Solder the jumpers underneath the microcontrollers

I could see how this could be easily missed though, since it's not part of the "front" or "back" list of things to do.

closed 2024-04-17 by raeedcho