raeedcho/temper #17: Building Miryoku does not work

Zoltan-Balazs, 2024-04-26:
Whenever I try to build the Miryoku firmware, it does not work, see: [https://github.com/Zoltan-Balazs/miryoku_zmk/actions/runs/8844616666](https://github.com/Zoltan-Balazs/miryoku_zmk/actions/runs/8844616666) 
I see that the Temper was added like the Corne to the Miryoku repository; building that in the exact same way, just changing `temper` to `corne` in the shield section is successful: [https://github.com/Zoltan-Balazs/miryoku_zmk/actions/runs/8844680219|https://github.com/Zoltan-Balazs/miryoku_zmk/actions/runs/8844680219]

calerouxz, 2024-04-26:
Try building both sides on separate runs. If I remember correctly, building them together resulted in errors for me.

Zoltan-Balazs, 2024-04-30:
Thank you, unfortunately the build still fails with `Error: Keymap found but shield not found. Board used for shield.` 
It seems like the workflow does not recognise `temper_right` or `temper_left` as valid shields

calerouxz, 2024-04-30:
I'll try a build later today and report back.

calerouxz, 2024-05-01:
This didn't build for me today either, it may be worth raising a discussion, as I can't see missing or otherwise missing information on the files.

calerouxz, 2024-05-02:
Update on this matter! I found the issue and why the build for the temper was broken on Miryoku. I'll keep this documented here for posterity and I'll try posting a PR to Miryoku ZMK. 
The outboard refs for temper needed to be updated, as [this commit](https://github.com/raeedcho/temper-zmk-config/commit/024290005a9ae565d3afd7ad716056e3e9947976) moved the files on the temper's repository and Miryoku outboards require the refs to build. I've opened [this PR](https://github.com/manna-harbour/miryoku_zmk/pull/51) to fix it in Miryoku ZMK.

raeedcho, 2024-05-02:
Ah yes, I had to move the files to comply with the new Zephyr update but didn't realize it would break the Miryoku build. Thanks for looking into and fixing it!

calerouxz, 2024-05-03:
The PR regarding the temper has been merged into Miryoku. I tested a build (and flashed) and it now works correctly. Manna also made a sample build which also compiled. @Zoltan-Balazs you can sync your fork and it should now work.

Zoltan-Balazs, 2024-05-06:
Thank you, it works as intended, I was able to verify on my own fork as well!
I don't think I would've been able to figure this out, thank you @calerouxz :smile:

closed 2024-05-06 by Zoltan-Balazs