raeedcho/temper #19: Reset switches hard to source

ThomasAlban, 2024-08-15:
Hi,
It’s practically impossible for me (uk) to source the reset button which is used on this keyboard without paying extortionate shipping. The one on the original chocofi (pts636) is much more easily available. Is there any chance of a version of this keyboard being made that supports that type of reset switch?
Thanks

raeedcho, 2024-08-16:
Ah, that is frustrating--I didn't realize that when designing the board. Unfortunately, I don't currently have plans to make a second version of the temper, but I wouldn't rule it out, if there are enough desired changes to make it worthwhile. Sorry! I'll leave this issue open for now as a feature request and let others comment.

ThomasAlban, 2024-08-16:
Yeah that would be great.
I'm really keen to build the temper as it seems like pretty much my ideal keyboard (I currently use a corne but I don't need the outer row of keys and I find the little finger stagger not low enough)
When I build it I'd like to add 1000mah batteries below the keyboard, so if I solder the battery connectors to the pads on the bottom of the pcb would it still work? (So I don't have to have wires protruding out of the keyboard)
If not, perhaps another modification could be a little indent above the battery pads so the wires can wrap around without protruding.

raeedcho, 2024-08-17:
Yes, you should be able to just solder the batteries onto the backside of the board. I would double check with a multimeter to make sure, but the PCB is designed so that the pads on front and back are directly connected.

AdrianBonpin, 2024-12-31:
Quick question, regarding the reset switch as I'm not proficient with reading pcbs (which I need to really learn). Will standard 4 pin reset switches work? Specifically these ones that I can source.
[image]

I apologize if my question has a simple answer or may be mistaken and I thank you in advance. (And an advanced Happy New Year)

raeedcho, 2025-01-02:
I'm sorry to say that I'm not sure this will fit in the same reset switch spot--it kind of looks like the contacts are not quite positioned correctly to sit on the PCB pads, but I can't be certain.

AdrianBonpin, 2025-01-03:
Footprint aside as I can simply play around with the placement, does it function the same? but just with 4 pads?

I was trying to trace where the 4 pads lead to but I really couldn't figure it out. Thanks again

anycancellable, 2025-05-13:
Hey, 
I just finished building (thanks @raeedcho) using a switch from [Aliexpress](https://vi.aliexpress.com/item/1005002845025014.html) (1005002845025014 item id).