raeedcho/temper #20: Just to say: Thank you !!

jfaixo, 2024-10-23:
I recently fell into the rabbit hole of the "diy keyboard" community, and picked your great design as a first keyboard to build :)

That went extremely well, "just" working from the first time! Dropping some pictures: https://imgur.com/a/s8EuhC4

[image]

Some little points that maybe could be better explained in the repository:

- PCB & electronic:
  - As I'm rather new (especially into electronics), I never saw a reversible PCB before. I had to look into kicad to understand how and on which side the triangle jumpers must be soldered. A word in the readme explaining that these jumpers must be soldered correctly depending on how the MCU will be oriented would be great!
  - everything went extremely smoothly, I followed [this guide](https://github.com/rafaeldelboni/buildlogs/blob/main/crkbd-choc-spaced-switch.md)
- case: usually there is a lot of focus on the electronic part, but little on the housing, and this is where I had a harder time to figure how to do it before ordering the parts
  - One trap I fell into (happily without consequence) is about the thickness of the switch plate. I just did not think to check before ordering. I used PCB based switch plates (the blue ones in the pictures), and ordered them with the default settings of jlcpcb. But the default PCB thickness is 1.6mm, and the thickness for Kailh choc switches to properly clip is...1.2mm. In the end this is not an issue, as the switches are already well engaged en the switch plate + also properly maintained by the PCB holes underneath. In the end there is no moving part at all, as everything is tightly engaged (even though I chose not to screw the switch plate. Screws are below it, completely hidden)
  - an other minor trap is that I did not see the option (enabled by default) that add a fabrication identifier on PCBs. For the switch plate this is a bit sad, as there is an undesired id visible on one of the two splits... (here again lucky me: choc spacing is very tight, this is almost invisible)
  - something I managed to figure out beforehand but is not explicited in the repository: the holes in the bottom plate provided have a diameter of 3.2mm, so normal M2 screws would pass through these holes... I chose to use screw inserts melted into the plexiglass. Also: I had to add a scale as a legend in the svg so that my local supplier can be sure of the dimensions of the end product

As a conclusion: THANK YOU VERY MUCH for the design! And if you wish I can push a PR to add the previous build details :)

raeedcho, 2024-10-25:
Nice, looks really great! I'm glad you were able to figure everything out. And thanks for the comments--I'll try to clarify things in the readme a bit more.

Just to clarify a few things:
- Regarding the jumper solder locations, the readme states to solder the jumpers on the front side of the PCB, directly under the microcontrollers. How should I reword this to clarify?
- That's strange about the bottom plate holes being too big for your M2 screws--they seemed to work totally fine for me. Which M2 screws did you use?

Thanks again for the constructive feedback! Hope you enjoy the new keyboard!

jfaixo, 2024-10-26:
> * Regarding the jumper solder locations, the readme states to solder the jumpers on the front side of the PCB, directly under the microcontrollers. How should I reword this to clarify?
Ok no sorry that's just me that did not carefully read everything... my bad ! (having to dig in to understand was also enjoyable !)

> * That's strange about the bottom plate holes being too big for your M2 screws--they seemed to work totally fine for me. Which M2 screws did you use?

I used:
- these inserts (M2 OD 3.5mm variant): https://fr.aliexpress.com/item/1005003582355741.html melted into plexiglass (3mm plexi)
- these spacers (2mm): https://fr.aliexpress.com/item/1005003068856952.html (because kailh hotswaps are 1.8mm tall)
- these (very standard) M.2 screws: https://fr.aliexpress.com/item/4000310491280.html in their 6mm variant

So: pcb 1.6mm + 2mm spacers + 3mm insert ~= 7mm, the M.2 screws nicely hold the whole thing :)

raeedcho, 2024-10-27:
I may not have totally understood your last comment about the backplate holes being too big--do you mean that the heads of the M2 screws would go through the holes? Or that the brass spacers I suggested in the repository would slide through?

If it's the latter, then I intentionally designed it so that the backplate would slide onto the spacers screwed onto the PCB. This means that the backplate isn't actually secured in place, though in practice, if the spacers are short enough, the plate is sandwiched quite well between the M2 screws heads and the sockets on the PCB, and it doesn't slide around.

If it's the former, I'm surprised because the M2 screws that I and others have used seem to have been the right size not to slip through and do indeed hold the backplate to the PCB. Though it's possible that not all M2 screws would work the same way.

Either way, I'm glad you found a good solution to the problem!

jfaixo, 2024-10-30:
Yes it is the latter option, standard M2 spacers just fit in the hole, so this is by design :) But my M2 screw heads barely grab the border of the hole (it's submillimetric). In the end I am very happy with the inserts melted in the backplate, it feels strong :muscle:

Your project inspired me, I might create a variation of it in 2025, for a more fixed setup, with underglow, bigger batteries, and a thumb cluster more fit to my hands (I have thin long fingers, I cannot reach the innermost thumb keys, without having to move away my palm).

Thanks again !

gurland, 2024-10-30:
Hello, @raeedcho, I`d like to address this:
>     * Regarding the jumper solder locations, the readme states to solder the jumpers on the front side of the PCB, directly under the microcontrollers. How should I reword this to clarify?

It was my first keyboard building experience and this step was not obvious at all. I spent a lot of time digging issues on GH and gerber file to understand what are these "jumpers".

So I`ve opened a PR #21 that clarifies what needs to be done in the picture

raeedcho, 2024-10-30:
Got it--thanks for the feedback and the PR!

@jfaixo Wonderful! Feel free to send me a link if and when you make your variation 😁

gurland, 2024-10-31:
@jfaixo could you please provde the svg file with the scale?

jfaixo, 2024-10-31:
Just created https://github.com/raeedcho/temper/pull/22 if you wish the scale as a first step.

But having it with "true" width/height information would be better I think. Like have 1 pixel == 1mm or 0.1mm and a proper width/height tagging seems to be the way.

raeedcho, 2024-10-31:
Ah, I think I misunderstood the initial problem with the SVG then. I didn't realize the scale wasn't already embedded in the SVG--when I open the file on Adobe Illustrator, it seems to have the scale correct. And when I sent the file to the laser cutter, it just sort of worked perfectly.

To clarify, @gurland, are you saying that the scale info that your manufacturer had when opening the file is wrong?

jfaixo, 2024-10-31:
Yes that was my issue: the first cost estimation I did with the original svg was not OK (hole size was 5.76mm according to their scaling interpretation).

gurland, 2024-10-31:
> To clarify, @gurland, are you saying that the scale info that your manufacturer had when opening the file is wrong?

Yes, after opening the original SVG file in their software it displayed 83.45mm X 67.493mm

raeedcho, 2024-11-06:
That's very strange--I tried uploading the file to [Ponoko](https://ponoko.com), and there, it says the file is 111.27 mm x 89.92 mm (see image).

[image]

What vendors did you two use? I'm wondering if maybe different vendors use different indicators for SVG sizing.