raeedcho/temper #27: Brass spacer

honweng, 2025-05-01:
Hello there, thanks for the awesome board. I have ordered the PCB print but have some doubts on the M2 brass standoff - may I know what is the maximum diameter of the brass standoff that I should be getting?

anycancellable, 2025-05-14:
@honweng Chipping in as I recently finished my build and also went through this question. I ordered my acrylic backplate from 42keebs, and Pete adjusted the file so the hole is ~2.2 mm which was a good fit for M2 screws and M2 female to female Brass standoffs.

closed 2025-05-14 by honweng

honweng, 2025-05-14:
@raeedcho : shouldn't the standoff diameter be bigger than 3.2cm, instead of smaller? A standoff smaller than 3.2cm will cause the back plate to slide through the standoff.

raeedcho, 2025-05-15:
The backplate is actually designed to slide over the standoffs and be secured in place with screws--this leads to a very low profile design.

raeedcho, 2025-05-15:
You are not wrong, but that is exactly how it was designed--the back plate slides on and the screws just fasten to the spacer. The back plate is sandwiched between the screw heads and the PCB (really the hot swap sockets), and in my boards, it doesn't slide along the spacers at all.

raeedcho, 2025-05-15:
> Ah, I didn't know that but I actually really like that idea. Could you specify that in the readme?
> 
> I thought the hole is used for the screws to go through. I even used a smaller standoff because with the 4mm standoff recommendation still felt too thick.

Ah, I didn't realize this was confusing--the assembly instructions currently read:

> - (optional) For back plate:
>   - attach standoffs to PCB with screws
>   - slide back plate onto standoffs and secure with screws

Do you have suggestions for how to re-word it?

anycancellable, 2025-05-15:
> (optional) For back plate:
> attach standoffs to PCB with screws
> slide back plate onto standoffs and secure with screws

> Do you have suggestions for how to re-word it?

How about: 

- (optional) For back plate:
    - attach standoffs to PCB with screws
    - align the back plate so the standoffs slide into the holes
    - fasten the back plate by screwing into the ends of the standoffs

I think _"into the holes"_ instead of _"onto standoffs"_ reads differently and probably makes it more clear.

raeedcho, 2025-05-19:
Thanks, I've updated the readme with a version of the new suggested language ("into holes" instead of "onto standoffs").