raeedcho/temper #28: Board/screen doesn't enter idle/deep sleep

anycancellable, 2025-05-13:
Hey! 

First of all, thanks @raeedcho for this awesome design, this was my first custom build (and first soldering) and it all turned out great. 
I added the optional nice!views and they work, except the keyboard seemingly doesn't enter idle/deep sleep mode. 

I didn't fork your config repo because I wanted to see how zmk installer works, but I copied your files to [my repo](https://github.com/anycancellable/zmk-temper). 

I'm not sure what's preventing the board to enter idle mode and let the displays turn off.

anycancellable, 2025-05-14:
Out of curiosity I installed a base corne firmware to test it, and both idle and deep sleep works, the keyboard turns off the display and disconnects from bluetooth so I guess either something gets overriden in the config or something is missing compared to corne. 

Does your board go to deep sleep without screens?

raeedcho, 2025-05-15:
I believe it does, but I suppose I haven't really tested the deep sleep function in a while (I usually just use the power switches to turn the boards off when I leave my computer or put the keyboards in my bag to take to work).

anycancellable, 2025-05-15:
I actually got it working after countless hours. The key was adding a `.conf` file under `/config`. I'm not exactly sure why this works. I have a different repo [here](https://github.com/anycancellable/zmk-config-temper), but it actually works for both. 

This repo that I linked, doesn't have `boards/shields/temper/boards/nice_nano_v2.overlay`. Can you tell me why was it needed for you?

closed 2025-05-18 by anycancellable

raeedcho, 2025-05-18:
Glad it's working for you now!

> This repo that I linked, doesn't have boards/shields/temper/boards/nice_nano_v2.overlay. Can you tell me why was it needed for you?

I'm not sure I remember why it was necessary before--it could be that I had started with a different version of ZMK possibly?